tongtybj/d_board #16: CANlib周辺の管理

chibi314, 2017-06-16:
Hydrusx_octoを作るにあたって通信の仕様変更をしたい。
その際にSpinal, Neuronで仕様を合わせるようにしないとまともに通信ができない（最悪、暴走する）ので、合わせるようにしたい。
とりあえずd_boardをsubmoduleとして入れればバージョン管理はできる。
追加としてCANlibは今2つのプロジェクトでほぼコピペになっているので、これ自体をsubmoduleとして入れるのが理想。ただし微妙に違うので分けられるようにしないといけない。

tongtybj, 2017-06-18:
@chibi314 

思ったんだけど、 neuron側のボードに固定ID(MACアドレスみたいな)をconst変数として振っておいて、
spinal(d_board)がCAN通信の初期にACKリクエストを送って、受け取ったneuronの数とID群とd_board自身が持っているuav_typeを基に、適切な通信内容(servo, motor, imu)を自動に決めるのはどうかな.

tongtybj, 2017-06-18:
@chibi314 

よくよく考えてみたら、
https://github.com/tongtybj/hydrus/blob/master/neuron/CAN/can_device.h#L20
が固定IDだね。。。

あんまり意味なかったかも。。。

tongtybj, 2017-06-25:
@chibi314 

前に話したneuron単位でのセンサデータ(imu+servo)の転送方法を起用すると、いろいろ変更が生じるのかな。

tongtybj, 2017-11-06:
@chibi314 

このissueは#30 と関連する議題になっているのかな。

chibi314, 2017-11-06:
@tongtybj 
そうです．

ただしCANlib自体はSpinal，Neuronで完全に一緒にすることはできないので，それはあきらめて，CAN通信で使う定数が書いてあるファイルを作成し，それを共通とするというやり方がよいと思いました．

そのファイルはすでに作ってあります．
https://github.com/tongtybj/d_board/blob/master/Jsk_Lib/CAN/can_constants.h

closed 2017-12-20 by tongtybj