tongtybj/d_board #27: /[joint,gimbal]_servo_config_cmdについての仕様検討

chibi314, 2017-08-16:
dragon branchにてCAN通信上のサーボの指令の仕様を以下のように変更した
15bit      torque enable
14bit      control enable
13-12bit reserved
11-0bit   joint angle

このtorque enableとcontrol enableについて正しく仕様を決めたい．今の仕様は以下の通り
* torque enable
このbitが変化するとサーボにtorque enableのinstruction packetが送られて，トルクオン/オフされる
* control enable
このbitが1の時は周期的にサーボに角度指令のinstruction packetを送る．0なら送らなくなる．

ここまでは設計として間違っていないと思う．初期状態はtorque_enable:1 control_enable:0である．問題はros上の/[joint,gimbal]_servo_config_cmdをどうするか．
実現したいこと→電源投入したらとりあえずその場でサーボオンされる．そのうちサーボオフしたり，角度指令を送れるようになる．
避けたいこと→サーボオフからサーボオンしたら前の角度指令が残っていたせいで新たにサーボオンしたときの姿勢で固まってくれない（場合によってはいきなり角度が変化して危険）

そこで今/[joint,gimbal]_servo_config_cmd(std_msgs/UInt8)は以下のような仕様
* TORQUE DISABLE(0)
torque_enable=0
control_enable=0
* TORQUE ENABLE(1)
torque_enable=1
control_enable=0
* CONTROL ENABLE(2)
torque_enable=1
control_enable=1

仕様として，どのコマンドにおいてもtorque_enableもしくはcontrol_enableの片方のみを操作することはせず両方とも操作するようにした．これは片方のみを操作すると今の状態を覚えておかないといけないためミスが起きると思ったので．（脳内ステートマシンが必要）
TORQUE_DISABLEはいわゆるサーボオフ，これにTORQUE_ENABLEを送ることでサーボをオンするが，角度指令は送られないため，その場でとどまることになる．手動でこのトピックにメッセージを送る場合はこれらのみを送るべきで，CONTROL_ENABLEを送るべきではない．CONTROL_ENABLEは安全に変形できる保証がある場合に送り，そこから角度を送るというように使うべきである．（地面にいるときは安全に変形できないので基本送らないほうがいい）

chibi314, 2017-11-03:
@tongtybj 
破棄で大丈夫です．
CANメッセージは元に戻しました．
15bit torque enable
14-12bit reserved
11-0bit joint angle

tongtybj, 2017-11-03:
確認できました。

closed 2017-11-03 by tongtybj