tongtybj/d_board #34: rosserialでrosservice serverが立てられない

chibi314, 2017-10-24:
@tongtybj 
題の通りなのですが，rosserial_stmはservice未対応ということでしょうか．
```
void testCallback(const std_srvs::Empty::Request & request, std_srvs::Empty::Response & response)
  {

  }
ros::ServiceServer<std_srvs::Empty::Request, std_srvs::Empty::Response> test_srv_("/test_service", testCallback);

nh_->advertiseService(test_srv_);
```
としましたが，PC側でrosservice listしても出てきませんでした．

PC側を起動したら，service callして機体の情報（slave数，サーボの数，それぞれのサーボのゲイン，etc）を一度PCに送って見れるようにするということが目的なのですが，topicを使って表現することも可能ですがserviceが使えるなら使うべきところだと思います．

tongtybj, 2017-10-25:
そうですね。まだ未対応ですね。
去年rosserial_stmを実装した時点では、
本家のrosserial_serverでもservice機能はちゃんと整っていなかったので、放置していました。
さっきみたら、本家の方でアップデータされていました。https://github.com/ros-drivers/rosserial/pull/208

落ち着いたら、service機能を実装します。
それまではtopicでやってください。

chibi314, 2017-10-25:
分かりました．時間があるときでいいのでお願いします．
サーボの設定（オーバーライト，ゲイン変更etc.）もserviceで書いた方が本当は正しそうですね．

tongtybj, 2017-12-20:
https://github.com/tongtybj/aerial_robot/pull/109 で対応

closed 2017-12-20 by tongtybj